KonradIT/gopro-py-api #129: Unable to set streaming window size on Hero 7 Black

MarkHargrove, 2020-01-11:
From a Python3 script on a Raspberry Pi 4 I'm able to view a stream from a GP Hero 7 Black using OpenCV.  The initialization code is:

`gpCam = GoProCamera.GoPro()
gpCam.overview()  # just to verify I'm connected

gpCam.video_settings(res='1080p', fps='30')
gpCam.gpControlSet(constants.Stream.WINDOW_SIZE, constants.Stream.WindowSize.R720)
gpCam.livestream("start")
cap = cv2.VideoCapture("udp://10.5.5.9:8554", cv2.CAP_FFMPEG)
`
...
(and then proceeding into the main loop using cap.read() calls)

Despite the request to use a 720p window, the stream is at 480p.

The code is running on an RPI Model 4 under a recent release of Buster.

Capricorn-3th, 2021-01-24:
I have the same problem,you have the solution? Thank you

MarkHargrove, 2021-06-19:
I never did find a solution for the above, but have moved on to a GPHero9 that doesn't have this issue.