KonradIT/gopro-py-api #131: Connect a second Go Pro with a second wifi adaptor

ldevozz, 2020-02-19:
Hi KonradIT,

Firstly thanks for your work on this API, I'm really enjoying using it for remote capture and transfer for computer vision projects.

I am currently trying to connect a second GP on a second wifi adaptor (wifi dongle). I have no trouble connecting to one gopro with it. I would like to do something like this:

gp1 = GoProCamera.GoPro()
gp2 = GoProCamera.GoPro()

So that I have two gopro object I can issue commands to. However when this is run gp1 and gp2 get setup to the same gopro that is connected to the first wifi adaptor. Is there something that can be passed to .GoPro() that can force it to look at the second wifi adaptor?

FYI I am using two Hero 5 Blacks, both have connected individually. Additionally I have had a play around with the pywifi package and have gotten to the interface function to give me status on the second wifi adaptor - I just haven't been able to link the second gopro.

Thanks in advance for any help.

Luke

Moogun, 2020-03-02:
Wonder if you have made any progress.

ldevozz, 2020-03-02:
@Moogun 
Unfortunately no, it would be really great if there was a solution to this. It seems that something should be possible as the .GoPro() method finds the second wifi adaptor (dongle) without help.  Even when the go pro is connected to that adaptor and the laptop wifi (set as the primary wifi) is already connected to a network.

ferrix, 2020-03-20:
Duplicate to #115

MaxwellNewberry, 2020-04-06:
@ldevozz Would you be willing to share you code you used with the pywifi library? I am currently working to do a similar project and have two GoPros connected to separate network interfaces but http://10.5.5.9 only works with one camera.