KonradIT/gopro-py-api #136: Video sizes are huge, causing slow download speeds over WiFi

MaxwellNewberry, 2020-04-05:
I am recording on a HERO 7 BLACK and getting huge video sizes at the lowest resolution settings. In return, trying to download these videos on a 5Ghz WiFi 5 connection still takes a long time. Are there any suggestions in the code or otherwise to reduce the video size or download speeds. I am controlling several different cameras and trying to download long videos.

KonradIT, 2020-04-05:
You could download the Low Res version. `downloadLowRes()`

MaxwellNewberry, 2020-04-05:
@KonradIT The purpose of the video is to annotate? Would the Low Res version lose enough integrity for the video not to work well with ML?

KonradIT, 2020-04-05:
LRV is usually 840x480 in size, at a low bitrate. Not the resolution or bitrate I'd use for machine learning.

MaxwellNewberry, 2020-04-06:
@KonradIT Thanks.

closed 2020-04-06 by MaxwellNewberry