KonradIT/gopro-py-api #151: GoPro Hero 8 Connect through cable?

chandranpranav, 2020-10-03:
Heyy guys I would really like some input here, I'm trying to connect my GoPro Hero 8 and use it as a camera on OpenCV applications. I'm using multiple cameras for object detection so cannot use WiFi for this approach as it can only detect one at a time.. When I connect my GoPro it is detected as a Network Adapter under the name 'GoPro RNDIS Device' I can use it on Skype, Zoom, and other applications but when I try to connect two cameras it detects only one camera.. But in device manager detects as two. What I want is to read these camera inputs through OpenCV. Any input would be appreciated. 
[image]

KonradIT, 2020-10-03:
Interesting, can you see the DHCP addresses for both cameras via ifconfig? Using the [experimental branch for webcam](https://github.com/KonradIT/gopro-py-api/pull/148) you could do:

`gopro1 = GoProCamera.GoPro(ip_address=GoProCamera.GoPro.getWebcamIP("IP_ADDRESS_CAM_1", camera=constants.gpcontrol, webcam_device="IP_ADDRESS_CAM_1")

gopro2 = GoProCamera.GoPro(ip_address=GoProCamera.GoPro.getWebcamIP("IP_ADDRESS_CAM_2", camera=constants.gpcontrol, webcam_device="IP_ADDRESS_CAM_2")

gopro1.startWebcam()
gopro2.startWebcam()`

Streams are on UDP on port 8554.

chandranpranav, 2020-10-14:
Hey @KonradIT so I'm using windows predominantly and I know they are still working in beta testing... But I tried to test it using the DHCP address . The script runs but always loads the laptop webcam instead. I would really like if you throw some light. 

this is my python code I've used for testing it out 

import cv2
from goprocam import GoProCamera, constants
gopro = GoProCamera.GoPro(ip_address="172.29.178.51") # change
#gopro = GoProCamera.GoPro(ip_address="172.29.194.51") # change
gopro.startWebcam(constants.Webcam.Resolution.R720p)
gopro.webcamFOV(constants.Webcam.FOV.Linear)
gopro.getWebcamPreview()
[image]

KonradIT, 2020-10-14:
Try using the v4l2 example from the examples folder.

KonradIT, 2020-10-19:
Can confirm I can use two Hero9 Black cameras over USB with GoPro Connect instead of MTP to take a photo: https://twitter.com/konrad_it/status/1318286019251392519

vukasinpetrovic, 2020-10-20:
@KonradIT Is this achievable with raspberry pi? I'm trying to keep camera always connected to PI in order to do timelapses, but I cannot figure out how to control it via USB.

KonradIT, 2020-10-20:
Hi, i still want to test a few stuff before I release to pip

edmondklai, 2020-12-16:
@KonradIT, I have a GoPro Max, and I am unable to connect through WiFi because it seems like it won't show up as a router when I turn on the wifi on the camera, it will only allow me to pair with the app.  I am trying to connect using a cable but the camera doesn't show up on the Network adapter, only on the Portable devices in Device Manager.  Do you know if there are settings I need to change on either my GoPro or laptop?

KonradIT, 2020-12-16:
No, GoPro MAX does not support USB Ethernet, only HERO8 and HERO9 do.

>I am unable to connect through WiFi because it seems like it won't show up as a router when I turn on the wifi on the camera, it will only allow me to pair with the app

GoPro MAX only starts the WiFi Access Point when you tell it to via Bluetooth, you can check how I do it [here](http://github.com/konradit/gopro-ble-py).

logburn, 2021-01-30:
Sorry to dig this up again, but are there any working code examples of using the Hero 8 with the cable instead of WiFi? I've tried to use the "normal" code and get nonrecoverable errors. Or is there perhaps a settings on the GoPro that needs to be changed? I have the USB connection set to GoPro Connect as specified above already

vukasinpetrovic, 2021-01-30:
> Sorry to dig this up again, but are there any working code examples of using the Hero 8 with the cable instead of WiFi? I've tried to use the "normal" code and get nonrecoverable errors. Or is there perhaps a settings on the GoPro that needs to be changed? I have the USB connection set to GoPro Connect as specified above already
> 
> edit: I just watched an example video and noticed I never get passed the `Waking up...` message without error. So I never get the `Camera successfully connected!` message

I have it on Raspberry but I can get to it only on Monday because I'm out of home for the weekend. I'll post it here when on Monday.

logburn, 2021-02-02:
@vukasinpetrovic Any update on this?

vukasinpetrovic, 2021-02-02:
`
#!/usr/bin/python3

from signal import signal, SIGINT
from goprocam import GoProCamera, constants

def handler(s, f):
    gopro.stopWebcam()
    quit()

signal(SIGINT, handler)

gopro = GoProCamera.GoPro(ip_address=GoProCamera.GoPro.getWebcamIP())
gopro.take_photo()
`

@logburn This is the code I used. One thing to note, when I was using this I think that functionality was on dev branch. Don't take my word for granted, it was 2-3 months ago, so check if it is still on dev or on master.

logburn, 2021-02-02:
Looks like it works on the current master branch. Thank you so much for getting that code for me!

wilbercl, 2022-03-25:
Hello, I am trying to use the **v4l2-hero9-webcam.py** script that you recommend in the comments above, and it does not work for me, it hangs. 
[image]

I have also tried to execute the following lines of python and it gives me an error with **vlc**, which if I have it installed.
`import cv2
from goprocam import GoProCamera, constants

gopro = GoProCamera.GoPro(ip_address="172.28.199.55") # change
#gopro = GoProCamera.GoPro(ip_address="172.29.194.51") # change
gopro.startWebcam(constants.Webcam.Resolution .R720p)
gopro.webcamFOV(constants.Webcam.FOV.Linear)
gopro.getWebcamPreview()`

[image]

I need to connect one or more GoPro as a webcam and then recognize them with OpenCV, thanks.

ethaniel, 2022-05-29:
> No, GoPro MAX does not support USB Ethernet, only HERO8 and HERO9 do.

HERO 8 and HERO 9 support USB Ethernet adapters?

shahla-ai, 2023-12-04:
@vukasinpetrovic 
hello i wish for your help
i am trying to control my **GoPro** via USB wired cable on my RPI4
i connected my cam to my PI, downloaded and installed version 4.1.0 of **gopro-pi-api**
is there anything i should do more , and what python code should i write to connect to my cam ?
best regards,

vukasinpetrovic, 2023-12-08:
@shahla-ai I did this years ago and I'm out of that scope now. I'm sorry that I'm unable to help you right now. Try to use examples of others and one that I provided in this issue (couple of comments above).