KonradIT/gopro-py-api #153: Missing zero padding in filenames when using downloadMultiShot()

TheKysek, 2020-11-14:
**Describe the bug**
Multishot sequences cannot be downloaded.

**To Reproduce**
Steps to reproduce the behavior:
1. Have file counter be less than 1000
2. Take a timelapse
3. Try to use `downloadMultiShot` to download it
4. See error

**Expected behavior**
Multishot gets downloaded

**Screenshots**

```
>>> g.downloadMultiShot(path='100GOPRO/G0040023.JPG')
filename: G00423.JPG
ERROR: HTTP Error 404: Not Found
filename: G00424.JPG
ERROR: HTTP Error 404: Not Found
filename: G00425.JPG
ERROR: HTTP Error 404: Not Found
filename: G00426.JPG
ERROR: HTTP Error 404: Not Found
filename: G00427.JPG
ERROR: HTTP Error 404: Not Found
filename: G00428.JPG
ERROR: HTTP Error 404: Not Found
filename: G00429.JPG
ERROR: HTTP Error 404: Not Found
filename: G00430.JPG
ERROR: HTTP Error 404: Not Found
```
As you can see, there is missing "00" between "G004" and "23.JPG"-"30.JPG".

The filenames should be:
- "G0040023.JPG"
- "G0040024.JPG"
...
- "G0040030.JPG"

**Additional context**

Bug in lines:

- https://github.com/KonradIT/gopro-py-api/blob/0249e073c725c1fe62d7a945fd0dcf9778b0c422/goprocam/GoProCamera.py#L839
- https://github.com/KonradIT/gopro-py-api/blob/0249e073c725c1fe62d7a945fd0dcf9778b0c422/goprocam/GoProCamera.py#L856

KonradIT, 2020-11-21:
True, will debug and push an update.

tecamenz, 2020-11-22:
Hi Konrad 

Thanks for your hard work and nice API!

I have a similar problem with where my SD-Card contains mixed recordings of normal photos and multishot sequences. 
The error is:
```Connected to 10.5.5.9
Traceback (most recent call last):
  File "/home/pi/gopro_timelapse/dump_sd.py", line 9, in <module>
    gpCam.downloadMultiShot()
  File "/home/pi/gopro_timelapse/.venv/lib/python3.7/site-packages/goprocam/GoProCamera.py", line 836, in downloadMultiShot
    lower_bound = i2["b"]
KeyError: 'b'
```

The multishot sequence is not the last recording. `downloadMultiShot()` querries the folder name and filename which in my case returns:
folder = "101GOPRO"
filename = "GOPR4987.JPG"
- https://github.com/KonradIT/gopro-py-api/blob/0249e073c725c1fe62d7a945fd0dcf9778b0c422/goprocam/GoProCamera.py#L825
- https://github.com/KonradIT/gopro-py-api/blob/0249e073c725c1fe62d7a945fd0dcf9778b0c422/goprocam/GoProCamera.py#L826

Manual inspection of the SD-Card shows that the multishot sequence is divided into two folders (100GOPRO and 101GOPRO)
For debug purpose, I attached the returned json "arr" from my camera and the complete folder structure of the SD-Card as a json file in the debug.zip at the end. 
https://github.com/KonradIT/gopro-py-api/blob/0249e073c725c1fe62d7a945fd0dcf9778b0c422/goprocam/GoProCamera.py#L827

[debug.zip](https://github.com/KonradIT/gopro-py-api/files/5580060/debug.zip)