KonradIT/gopro-py-api #156: Traceback (most recent call last):   File "./gopro-py-api/examples/launch_webcam_preview.py", line 6,

lukiifaz, 2020-11-20:
Traceback (most recent call last):
  File "./gopro-py-api/examples/launch_webcam_preview.py", line 6, in <module>
    sys.argv[1]), camera=constants.gpcontrol, webcam_device=sys.argv[1])
IndexError: list index out of range

KonradIT, 2020-11-20:
You're supposed to put your interface as an argument.

lukiifaz, 2020-11-20:
Rpi3b + with octoprint, hero5 sesion is conected to rpi via wifi, rpi is conected to internet via ethernet. You mean network interface? Any example?

KonradIT, 2020-11-21:
>hero5 sesion is conected to rpi via wifi

Why are you trying to use the webcam example (which uses GoPro Connect functionality only found in Hero8 and 9 cameras) for the Hero5 Session?

lukiifaz, 2020-11-21:
Because I didn't know it. So how do I preview octoprint. I have Domoticz and therefore I am looking for a solution with displaying the image in the browser.

KonradIT, 2020-11-21:
You'll have to figure that out yourself since I don't have octoprint, or a 3d printer. There are tutorials out there for how to integrate GoPro into OctoPrint via OctoLapse.

closed 2020-11-21 by KonradIT