KonradIT/gopro-py-api #161: Unable to fetch most recent media URL on GoPro Fusion

sahajamatya, 2020-12-23:
- OS: macOS 11.1 (Big Sur)
 - GoPro: Fusion

I get an almost empty file path when using the getMedia() and getMediaFusion() functions. 

**To Reproduce**
Steps to reproduce the behavior:
```
>>> from goprocam import GoProCamera, constants
>>> GoProCamera.GoPro().getMediaFusion()
```

**Expected behavior**
It should return the URL for both front and back images of last picture taken. 

`['http://10.5.5.9/videos/DCIM/100GBACK/GPBK0151.JPG', 'http://10.5.5.9/videos2/DCIM/100GFRNT/GPFR0151.JPG']`

**Screenshots**
Instead, it returns: 

`['http://10.5.5.9/videos/DCIM//', 'http://10.5.5.9/videos2/DCIM//']`

[image]

**Additional context**
The SD cards are not empty. 

I am able to view images by manually typing in the correct URL on the browser. 

goprocam/GoProCamera.py lines 649-677? Are "GBACK", "GFRNT" supposed to be prefixed with "100" to be "100GBACK", "100GFRNT"?

KonradIT, 2020-12-24:
Hi, can you rollback to this version? https://github.com/KonradIT/gopro-py-api/releases/tag/3.0.1 and try again

sahajamatya, 2020-12-24:
Thank you for your response, Konrad. It seems I have the same issue with v3.0.1 as well.

[image]