KonradIT/gopro-py-api #162: Fetching real time accelerometer data for Fusion

sahajamatya, 2020-12-23:
Can accelerometer data be fetched in real time for Fusion? I am trying to automate the image taking process so that the interval between each picture taken would depend upon what speed I am traveling at.

HenryKang1, 2020-12-28:
Did you figure it out? I am also wondering about it.

sahajamatya, 2020-12-28:
HenryKang1,

I pivoted my application to be mobile web-based, this has allowed me to utilize the phone's accelerometer with JavaScript instead of relying on data from the GoPro which would've been nice. I am not aware if something similar can be done with python. [Here](https://youtu.be/06MV79PPMEo) is a YouTube video I used as a reference.

HenryKang1, 2020-12-28:
> HenryKang1,
>
> I pivoted my application to be mobile web-based, this has allowed me to utilize the phone's accelerometer with JavaScript instead of relying on data from the GoPro which would've been [...]

Thank you for application. I want to apply Gopro for VIO so I just want to check gopro provide the imu data wirelessly.