KonradIT/gopro-py-api #170: Hero 3 error on take photo

pgregg88, 2021-03-21:
I get an error upon attempting to take a photo with a Hero 3 Black.
I've seen others post similar issues. I've tried everything in those posts and I still get the same result
I've also manually replaced the GoProCamera.py with the version published on Dec 8, 2020. Same result

**To Reproduce**
>>> from goprocam import GoProCamera
>>> from goprocam import constants
>>> gpCam = GoProCamera.GoPro(constants.auth)
Connected to 10.5.5.9
>>> gpCam.take_photo()
bx00x02x04x0cHD3030300x13Hero3BlackEdition
bx00x02x04x0cHD3030300x13Hero3BlackEdition
bx00x02x04x0cHD3030300x13Hero3BlackEdition
Traceback (most recent call last):
  File "/usr/lib/python3.6/http/client.py", line 297, in _read_status
    status = int(status)
ValueError: invalid literal for int() with base 10: 'ÿûAllow:'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/ubuntu/.virtualenvs/goprocam/lib/python3.6/site-packages/goprocam/GoProCamera.py", line 574, in take_photo
    return self.getMedia()
  File "/home/ubuntu/.virtualenvs/goprocam/lib/python3.6/site-packages/goprocam/GoProCamera.py", line 636, in getMedia
    raw_data = self._request("gp/gpMediaList")
  File "/home/ubuntu/.virtualenvs/goprocam/lib/python3.6/site-packages/goprocam/GoProCamera.py", line 149, in _request
    return urllib.request.urlopen(uri, timeout=_timeout, context=_context).read()
  File "/usr/lib/python3.6/urllib/request.py", line 223, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.6/urllib/request.py", line 526, in open
    response = self._open(req, data)
  File "/usr/lib/python3.6/urllib/request.py", line 544, in _open
    '_open', req)
  File "/usr/lib/python3.6/urllib/request.py", line 504, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.6/urllib/request.py", line 1353, in http_open
    return self.do_open(http.client.HTTPConnection, req)
  File "/usr/lib/python3.6/urllib/request.py", line 1328, in do_open
    r = h.getresponse()
  File "/usr/lib/python3.6/http/client.py", line 1373, in getresponse
    response.begin()
  File "/usr/lib/python3.6/http/client.py", line 311, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python3.6/http/client.py", line 301, in _read_status
    raise BadStatusLine(line)

NMiklave, 2021-03-22:
I get the same exact traceback with a 3+ silver running a raspberry pi. The same valueError is raised when calling downloadLastMedia,

pgregg88, 2021-03-26:
@NMiklave Any progress? I checked #107, but it seems unrelated. Were you referencing a difference issue by chance? Thx.

KonradIT, 2021-03-27:
Hi, that seems like a bug in the `requests` library... GoPro HERO3 server isn't returning a correct status error. Unfortunately my last Hero3 camera broke a few years ago so I cannot test it. What version of `requests` are you using?

pgregg88, 2021-03-27:
@KonradIT Hello. I'm running Python 3.6.9. Think I should upgrade to Python 3.9.2?

KonradIT, 2021-03-27:
To test, run `urllib.request.urlopen("http://10.5.5.9/gp/gpMediaList")` on both Python 3.6 and 3.9 and see if on 3.9 it still fails.

pgregg88, 2021-03-28:
@KonradIT I get the same results with I run the urllib test in Python 3.6 and 3.9.

Here's the output:
3.6.9
`(goprocam) ubuntu@ubuntu:~$ python
Python 3.6.9 (default, Jan 26 2021, 15:33:00) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import urllib.request
>>> urllib.request.urlopen("http://10.5.5.9/gp/gpMediaList")
Traceback (most recent call last):
  File "/usr/lib/python3.6/http/client.py", line 297, in _read_status
    status = int(status)
ValueError: invalid literal for int() with base 10: 'ÿûAllow:'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.6/urllib/request.py", line 223, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.6/urllib/request.py", line 526, in open
    response = self._open(req, data)
  File "/usr/lib/python3.6/urllib/request.py", line 544, in _open
    '_open', req)
  File "/usr/lib/python3.6/urllib/request.py", line 504, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.6/urllib/request.py", line 1353, in http_open
    return self.do_open(http.client.HTTPConnection, req)
  File "/usr/lib/python3.6/urllib/request.py", line 1328, in do_open
    r = h.getresponse()
  File "/usr/lib/python3.6/http/client.py", line 1373, in getresponse
    response.begin()
  File "/usr/lib/python3.6/http/client.py", line 311, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python3.6/http/client.py", line 301, in _read_status
    raise BadStatusLine(line)
http.client.BadStatusLine: HTTP/1.1 ÿûAllow: GET 

>>> 
`

3.9.2

`(goprocam_3.9.2) ubuntu@ubuntu:~$ python3.9
Python 3.9.2 (default, Feb 20 2021, 20:57:50) 
[GCC 7.5.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import urllib.request
>>> urllib.request.urlopen("http://10.5.5.9/gp/gpMediaList")
Traceback (most recent call last):
  File "/usr/lib/python3.9/http/client.py", line 293, in _read_status
    status = int(status)
ValueError: invalid literal for int() with base 10: 'ÿûAllow:'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3.9/urllib/request.py", line 214, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.9/urllib/request.py", line 517, in open
    response = self._open(req, data)
  File "/usr/lib/python3.9/urllib/request.py", line 534, in _open
    result = self._call_chain(self.handle_open, protocol, protocol +
  File "/usr/lib/python3.9/urllib/request.py", line 494, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.9/urllib/request.py", line 1375, in http_open
    return self.do_open(http.client.HTTPConnection, req)
  File "/usr/lib/python3.9/urllib/request.py", line 1350, in do_open
    r = h.getresponse()
  File "/usr/lib/python3.9/http/client.py", line 1347, in getresponse
    response.begin()
  File "/usr/lib/python3.9/http/client.py", line 307, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python3.9/http/client.py", line 297, in _read_status
    raise BadStatusLine(line)
http.client.BadStatusLine: HTTP/1.1 ÿûAllow: GET 

>>> `

pgregg88, 2021-03-28:
@KonradIT Could this error be authentication related? Based on the examples you provide, it appears the Hero3 authenticates differently than the later models. Just a thought. Thx.

KonradIT, 2021-03-28:
Hi, `/gp/gpMediaList` isnt' authenticated with the password... Can you browse the URL in your browser? This is weird because I tested [with Hero3 Black and getting last media worked](https://vimeo.com/209181246) (01:33 timestamp).

pgregg88, 2021-03-28:
@KonradIT I think I discovered something. I was able to get results if I added ":8080" to the URL.

`Python 3.6.9 (default, Jan 26 2021, 15:33:00) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import urllib.request
>>> with urllib.request.urlopen('http://10.5.5.9:8080/gp/gpMediaList') as f:
...      print(f.read(900))
... 
b'{"id":"3348086306992407555","media":[{"d":"100GOPRO","fs":[{"n":"GOPR0171.JPG","s":"2208880"},{"n":"GOPR0172.JPG","s":"2101300"},{"n":"GOPR0173.JPG","s":"2370476"},{"n":"GOPR0174.JPG","s":"2391124"}]}]}\n'
>>> `

KonradIT, 2021-03-28:
Ah, interesting, can you try my API with this constructor:

    gopro = GoProCamera.GoPro(ip_address="10.5.5.9:8080", camera=constants.auth)

pgregg88, 2021-03-28:
Progress for sure. It appears to be connecting, but throws a 404 error. I haven't been able to get it to actually take a picture. Will test more.

`Python 3.6.9 (default, Jan 26 2021, 15:33:00) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from goprocam import GoProCamera
>>> from goprocam import constants
>>> gpCam = GoProCamera.GoPro(ip_address="10.5.5.9:8080", camera=constants.auth)
Error code:404
Make sure the connection to the WiFi camera is still active.
Connected to 10.5.5.9:8080
>>> print(gpCam.take_photo(10))
Error code:404
Make sure the connection to the WiFi camera is still active.
wait 10 seconds.
Error code:404
Make sure the connection to the WiFi camera is still active.
http://10.5.5.9:8080/videos/DCIM/100GOPRO/GOPR0174.JPG
>>> `

NMiklave, 2021-03-28:
Same story on my end so far. couldn't get to the URL in my browser until I added the :8080. I can navigate the filesystem from the camera IP in my browser. Any command now throws the 404 error. @pgregg88, have you not been able to take a picture at all, or just with this change? My camera has been taking pictures, but the code crashes immediately afterwards. 

I tried updating the requests library from 2.18 to 2.25 and the requests-unixsocket from 0.1.5 to 0.2.0. Neither did the trick.

KonradIT, 2021-03-28:
You could try applying this patch to `GoProCamera.py`:

````diff
diff --git a/goprocam/GoProCamera.py b/goprocam/GoProCamera.py
index 01ffa1e..4255b83 100644
--- a/goprocam/GoProCamera.py
+++ b/goprocam/GoProCamera.py
@@ -144,7 +144,7 @@ class GoPro:
                                      self.ip_addr, path, param, value)
         elif param == "" and value == "":
             uri = "%s%s/%s" % ("https://" if _isHTTPS else "http://",
-                               self.ip_addr, path)
+                               self.ip_addr + ":8080" if path == "gp/gpMediaList" and self._camera == constants.Camera.Interface.Auth else "", path)
         if self._camera == constants.Camera.Interface.Auth:
             return urllib.request.urlopen(uri, timeout=_timeout, context=_context).read()
         else:
````

NMiklave, 2021-03-29:
Applied the patch, throws a new error when calling `gpCam = GoProCamera.GoPro(constants.auth)` and with the other call `gpCam = GoProCamera.GoPro(ip_address="10.5.5.9:8080", camera=constants.auth)`. Heres the new error.

Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/goprocam/GoProCamera.py", line 213, in sendBacpac
    self.getPassword() + value_notempty)
  File "/usr/local/lib/python3.6/dist-packages/goprocam/GoProCamera.py", line 135, in _request
    return urllib.request.urlopen(uri, timeout=_timeout, context=_context).read()
  File "/usr/lib/python3.6/urllib/request.py", line 223, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.6/urllib/request.py", line 524, in open
    req = meth(req)
  File "/usr/lib/python3.6/urllib/request.py", line 1248, in do_request_
    raise URLError('no host given')
urllib.error.URLError: <urlopen error no host given>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/nicholas/.config/JetBrains/PyCharmCE2020.3/scratches/scratch_4.py", line 7, in <module>
    gpCam = GoProCamera.GoPro(ip_address="10.5.5.9", camera=constants.auth)
  File "/usr/local/lib/python3.6/dist-packages/goprocam/GoProCamera.py", line 65, in __init__
    self.power_on_auth()
  File "/usr/local/lib/python3.6/dist-packages/goprocam/GoProCamera.py", line 490, in power_on_auth
    return self.sendBacpac("PW", "01")
  File "/usr/local/lib/python3.6/dist-packages/goprocam/GoProCamera.py", line 215, in sendBacpac
    print("Error code:" + str(error.code) +
AttributeError: 'URLError' object has no attribute 'code'

Process finished with exit code 1

KonradIT, 2021-03-29:
Try without the `ip_address="10.5.5.9:8080"` part. From what I'm seeing, the 8080 port is needed only on gp/gpMediaList, the rest uses the standard 80 port.

pgregg88, 2021-03-29:
I'm getting a similar error using:

gopro = GoProCamera.GoPro(constants.auth)

@KonradIT thank you for all of the help on this.

'(goprocam_3.9) pgregg@wewx:~/goprocam$ python
Python 3.8.5 (default, Jan 27 2021, 15:41:15) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from goprocam import GoProCamera, constants
>>> gopro = GoProCamera.GoPro(constants.auth)
Traceback (most recent call last):
  File "/home/pgregg/.virtualenvs/goprocam_3.9/lib/python3.8/site-packages/goprocam/GoProCamera.py", line 213, in sendBacpac
    return self._request("bacpac/" + param + "?t=" +
  File "/home/pgregg/.virtualenvs/goprocam_3.9/lib/python3.8/site-packages/goprocam/GoProCamera.py", line 135, in _request
    return urllib.request.urlopen(uri, timeout=_timeout, context=_context).read()
  File "/usr/lib/python3.8/urllib/request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.8/urllib/request.py", line 522, in open
    req = meth(req)
  File "/usr/lib/python3.8/urllib/request.py", line 1274, in do_request_
    raise URLError('no host given')
urllib.error.URLError: <urlopen error no host given>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pgregg/.virtualenvs/goprocam_3.9/lib/python3.8/site-packages/goprocam/GoProCamera.py", line 65, in __init__
    self.power_on_auth()
  File "/home/pgregg/.virtualenvs/goprocam_3.9/lib/python3.8/site-packages/goprocam/GoProCamera.py", line 491, in power_on_auth
    return self.sendBacpac("PW", "01")
  File "/home/pgregg/.virtualenvs/goprocam_3.9/lib/python3.8/site-packages/goprocam/GoProCamera.py", line 216, in sendBacpac
    print("Error code:" + str(error.code) +
AttributeError: 'URLError' object has no attribute 'code''

KonradIT, 2021-03-29:
Ok, maybe lets change `error.code` to `error.reason` since both `URLError` and `HTTPError` have those.

KonradIT, 2021-03-29:
Just out of curiosity are you on the latest firmware?

pgregg88, 2021-03-30:
@KonradIT I'm on 4.1.0. testing error.reason now.

pgregg88, 2021-03-30:
@KonradIT @NMiklave error.reason test

'Python 3.9.2 (default, Feb 20 2021, 20:56:08) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> sys.path.append(r'/home/pgregg/.virtualenvs/goprocam_3.9/lib/python3.8/site-packages')
>>> from goprocam import GoProCamera, constants
>>> gopro = GoProCamera.GoPro(constants.auth)
Error code:no host given
Make sure the connection to the WiFi camera is still active.
Connected to 10.5.5.9
>>> gopro.take_photo()
Error code:no host given
Make sure the connection to the WiFi camera is still active.
Error code:no host given
Make sure the connection to the WiFi camera is still active.
'http://10.5.5.9/videos/DCIM/100GOPRO/GOPR0176.JPG'
>>> '

pgregg88, 2021-04-01:
@KonradIT @NMiklave **SUCCESS!**

`Python 3.9.2 (default, Feb 20 2021, 20:56:08) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> sys.path.append(r'/home/pgregg/.virtualenvs/goprocam_3.9/lib/python3.8/site-packages')
>>> from goprocam import GoProCamera, constants
>>> gopro = GoProCamera.GoPro(camera=constants.auth)
Connected to 10.5.5.9
>>> gopro.take_photo()
bx00x02x04x0cHD3030300x13Hero3BlackEdition
bx00x02x04x0cHD3030300x13Hero3BlackEdition
bx00x02x04x0cHD3030300x13Hero3BlackEdition
'http://10.5.5.9/videos/DCIM/100GOPRO/GOPR0177.JPG'
>>> gopro.getMedia()
bx00x02x04x0cHD3030300x13Hero3BlackEdition
'http://10.5.5.9/videos/DCIM/100GOPRO/GOPR0177.JPG'
>>> gopro.shutter(constants.start)
b'\x00\x01'
>>> gopro.shutter(constants.stop)
b'\x00\x01'
>>> 
`

@KonradIT I really appreciate you working with us to get this figured out. Thank you very much.

KonradIT, 2021-04-01:
Ah, so the patch works? I'll push it to releases/pypi. So weird considering I tested with the Hero3 Black when I had it...

pgregg88, 2021-04-01:
@KonradIT The second patch does the trick. Thank you.

Just noticed I'm getting a similar error when I run 
' gopro.downloadLastMedia()'

'>>> gopro.downloadLastMedia()
bx00x02x04x0cHD3030300x13Hero3BlackEdition
bx00x02x04x0cHD3030300x13Hero3BlackEdition
filename: GOPR0180.JPG
size: 2.23MB
bx00x02x04x0cHD3030300x13Hero3BlackEdition
bx00x02x04x0cHD3030300x13Hero3BlackEdition
bx00x02x04x0cHD3030300x13Hero3BlackEdition
Traceback (most recent call last):
  File "/usr/lib/python3.9/http/client.py", line 293, in _read_status
    status = int(status)
ValueError: invalid literal for int() with base 10: 'ÿûAllow:'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pgregg/.virtualenvs/goprocam_3.9/lib/python3.8/site-packages/goprocam/GoProCamera.py", line 879, in downloadLastMedia
    urllib.request.urlretrieve(
  File "/usr/lib/python3.9/urllib/request.py", line 239, in urlretrieve
    with contextlib.closing(urlopen(url, data)) as fp:
  File "/usr/lib/python3.9/urllib/request.py", line 214, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python3.9/urllib/request.py", line 517, in open
    response = self._open(req, data)
  File "/usr/lib/python3.9/urllib/request.py", line 534, in _open
    result = self._call_chain(self.handle_open, protocol, protocol +
  File "/usr/lib/python3.9/urllib/request.py", line 494, in _call_chain
    result = func(*args)
  File "/usr/lib/python3.9/urllib/request.py", line 1375, in http_open
    return self.do_open(http.client.HTTPConnection, req)
  File "/usr/lib/python3.9/urllib/request.py", line 1350, in do_open
    r = h.getresponse()
  File "/usr/lib/python3.9/http/client.py", line 1347, in getresponse
    response.begin()
  File "/usr/lib/python3.9/http/client.py", line 307, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python3.9/http/client.py", line 297, in _read_status
    raise BadStatusLine(line)
http.client.BadStatusLine: HTTP/1.1 ÿûAllow: GET '

Should I assume a similar patch in the right place would resolve this as well? If so, I'll try to fix this one myself. I'll also test the other functions. Thx.

KonradIT, 2021-04-01:
If you try rolling back to, say, Python 3.4 and use the same code, maybe that actually gets rid of the bug, seems to be a core bug since it's on the `http` core library.

pgregg88, 2021-04-06:
No luck rolling back. I tried 3.4, 3.0, 3.6 and 3.9. It appears to be something related to the Hero3 response that sets off python. I get the same errors using requests.

I'm able to pull files using wget.download('http://10.5.5.9:8080/videos/DCIM/100GOPRO/GOPR0126.JPG',current_folder) so I have enough working to cover my immediate needs. Thanks for the help.

pgregg88, 2021-04-07:
@KonradIT @NMiklave 

After some more investigation, it appears to be related to not having the :8080 added to the URL for the download functions.

Here's my test code:

```
import requests
from requests.exceptions import HTTPError

for url in ['http://10.5.5.9/videos/DCIM/100GOPRO/', 'http://10.5.5.9:8080/videos/DCIM/100GOPRO/G0031250.JPG']:
    try:
        response = requests.get(url)

        # If the response was successful, no Exception will be raised
        response.raise_for_status()
    except HTTPError as http_err:
        print(f'HTTP error occurred: {http_err}')  # Python 3.6
    except Exception as err:
        print(f'Other error occurred: {err}')  # Python 3.6
    else:
        print('Success!')
```

Here's my output:

```
$ python3.9 2.py
Other error occurred: ('Connection aborted.', BadStatusLine('HTTP/1.1 ÿûAllow: GET \r\n'))
Success!
```

Here's an example of some errors that may mean more to you than they do to me.

```
>>> gopro = GoProCamera.GoPro(camera=constants.auth)
Connected to 10.5.5.9
>>> gopro.downloadAll()
filename: G0011210.JPG
bx00x02x04x0cHD3030300x13Hero3BlackEdition
Traceback (most recent call last):
  File "/home/pgregg/.pyenv/versions/3.9.2/lib/python3.9/http/client.py", line 293, in _read_status
    status = int(status)
ValueError: invalid literal for int() with base 10: 'ÿûAllow:'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pgregg/.pyenv/versions/3.9.2/lib/python3.9/site-packages/goprocam/GoProCamera.py", line 973, in downloadAll
    self.downloadMedia(folder, file, folder+"-"+file)
  File "/home/pgregg/.pyenv/versions/3.9.2/lib/python3.9/site-packages/goprocam/GoProCamera.py", line 931, in downloadMedia
    urllib.request.urlretrieve(
  File "/home/pgregg/.pyenv/versions/3.9.2/lib/python3.9/urllib/request.py", line 239, in urlretrieve
    with contextlib.closing(urlopen(url, data)) as fp:
  File "/home/pgregg/.pyenv/versions/3.9.2/lib/python3.9/urllib/request.py", line 214, in urlopen
    return opener.open(url, data, timeout)
  File "/home/pgregg/.pyenv/versions/3.9.2/lib/python3.9/urllib/request.py", line 517, in open
    response = self._open(req, data)
  File "/home/pgregg/.pyenv/versions/3.9.2/lib/python3.9/urllib/request.py", line 534, in _open
    result = self._call_chain(self.handle_open, protocol, protocol +
  File "/home/pgregg/.pyenv/versions/3.9.2/lib/python3.9/urllib/request.py", line 494, in _call_chain
    result = func(*args)
  File "/home/pgregg/.pyenv/versions/3.9.2/lib/python3.9/urllib/request.py", line 1375, in http_open
    return self.do_open(http.client.HTTPConnection, req)
  File "/home/pgregg/.pyenv/versions/3.9.2/lib/python3.9/urllib/request.py", line 1350, in do_open
    r = h.getresponse()
  File "/home/pgregg/.pyenv/versions/3.9.2/lib/python3.9/http/client.py", line 1347, in getresponse
    response.begin()
  File "/home/pgregg/.pyenv/versions/3.9.2/lib/python3.9/http/client.py", line 307, in begin
[... 127 lines cut ...]
GOPR1219.JPG
G0021220.JPG
G0031250.JPG
GOPR1280.JPG
G0011210.JPG
GOPR1211.JPG
GOPR1212.JPG
GOPR1213.JPG
GOPR1214.JPG
GOPR1215.JPG
GOPR1216.JPG
GOPR1217.JPG
GOPR1218.JPG
GOPR1219.JPG
G0021220.JPG
G0031250.JPG
GOPR1280.JPG
>>> gopro.listMedia(True, True)
bx00x02x04x0cHD3030300x13Hero3BlackEdition
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pgregg/.pyenv/versions/3.9.2/lib/python3.9/site-packages/goprocam/GoProCamera.py", line 754, in listMedia
    [i["d"], i2["n"], i2["s"], i2["mod"]])
KeyError: 'mod'
```

pgregg88, 2021-04-13:
Ordered a Hero9 :-)

KonradIT, 2021-04-13:
Nice! You'll notice this API will not be able to do a bunch of stuff over WiFi (eg: change modes, start record...) so check [gopro-ble-py](https://github.com/konradit/gopro-ble-py) for a Bluetooth implementation of the GoPro API.