KonradIT/gopro-py-api #171: Cannot stream GoPro Hero 8 Black video ...

simspace-dev, 2021-03-31:
Hi, 

I am trying to stream GoPro Hero 8 Black video to MPV as shown in this video you created called 
"GoPro HERO7 with Python API":  https://www.youtube.com/watch?v=i-X4fPVfoW0

I am on Mac OS Catalina running Python 3.9.

I can start video using `gopro.shoot_video(10)` and shoot a picture using `gopro.take_photo()`.

But `gopro.overview()` throws an error and `gopro.stream("udp://127.0.0.1:10000")` displays "packet corrupt" messages. 

Below is my CLI log to show you what is happening.

Thanks,
Chris

```
>>> gopro.stream("udp://127.0.0.1:10000")
ffmpeg version 4.3.2 Copyright (c) 2000-2021 the FFmpeg developers
  built with Apple clang version 12.0.0 (clang-1200.0.32.29)
  configuration: --prefix=/usr/local/Cellar/ffmpeg/4.3.2_3 --enable-shared --enable-pthreads --enable-version3 --enable-avresample --cc=clang --host-cflags= --host-ldflags= --enable-ffplay --enable-gnutls --enable-gpl --enable-libaom --enable-libbluray --enable-libdav1d --enable-libmp3lame --enable-libopus --enable-librav1e --enable-librubberband --enable-libsnappy --enable-libsrt --enable-libtesseract --enable-libtheora --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxml2 --enable-libxvid --enable-lzma --enable-libfontconfig --enable-libfreetype --enable-frei0r --enable-libass --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-libspeex --enable-libsoxr --enable-libzmq --enable-libzimg --disable-libjack --disable-indev=jack --enable-videotoolbox
  libavutil      56. 51.100 / 56. 51.100
  libavcodec     58. 91.100 / 58. 91.100
  libavformat    58. 45.100 / 58. 45.100
  libavdevice    58. 10.100 / 58. 10.100
  libavfilter     7. 85.100 /  7. 85.100
  libavresample   4.  0.  0 /  4.  0.  0
  libswscale      5.  7.100 /  5.  7.100
  libswresample   3.  7.100 /  3.  7.100
  libpostproc    55.  7.100 / 55.  7.100
[mpegts @ 0x7fe80a810a00] Packet corrupt (stream = 0, dts = 0).
[mpegts @ 0x7fe80a810a00] Packet corrupt (stream = 0, dts = 24024).
[h264 @ 0x7fe80b011800] left block unavailable for requested intra4x4 mode -1
[h264 @ 0x7fe80b011800] error while decoding MB 0 2, bytestream 13821
[h264 @ 0x7fe80b011800] concealing 1533 DC, 1533 AC, 1533 MV errors in I frame
[mpegts @ 0x7fe80a810a00] Packet corrupt (stream = 0, dts = 159159).
[h264 @ 0x7fe80b011800] concealing 146 DC, 146 AC, 146 MV errors in P frame
[mpegts @ 0x7fe80a810a00] Packet corrupt (stream = 0, dts = 414414).
[h264 @ 0x7fe80b011800] concealing 394 DC, 394 AC, 394 MV errors in P frame
[mpegts @ 0x7fe80a810a00] Packet corrupt (stream = 0, dts = 441441).
[h264 @ 0x7fe80b011800] cabac decode of qscale diff failed at 27 27
[h264 @ 0x7fe80b011800] error while decoding MB 27 27, bytestream 13961
[h264 @ 0x7fe80b011800] concealing 181 DC, 181 AC, 181 MV errors in P frame
[mpegts @ 0x7fe80a810a00] Packet corrupt (stream = 0, dts = 585585).
[h264 @ 0x7fe80b011800] concealing 52 DC, 52 AC, 52 MV errors in P frame
[mpegts @ 0x7fe80a810a00] Packet corrupt (stream = 0, dts = 630630).
[mpegts @ 0x7fe80a810a00] Packet corrupt (stream = 0, dts = 717717).
[h264 @ 0x7fe80b011800] left block unavailable for requested intra4x4 mode -1
[h264 @ 0x7fe80b011800] error while decoding MB 0 2, bytestream 36961
[h264 @ 0x7fe80b011800] concealing 1533 DC, 1533 AC, 1533 MV errors in I frame
[mpegts @ 0x7fe80a810a00] Packet corrupt (stream = 0, dts = 801801).
[h264 @ 0x7fe80b011800] concealing 351 DC, 351 AC, 351 MV errors in P frame
[mpegts @ 0x7fe80a810a00] Packet corrupt (stream = 0, dts = 1021020).
[h264 @ 0x7fe80b011800] concealing 315 DC, 315 AC, 315 MV errors in P frame
[mpegts @ 0x7fe80a810a00] Packet corrupt (stream = 0, dts = 1174173).
[h264 @ 0x7fe80b011800] concealing 1401 DC, 1401 AC, 1401 MV errors in I frame
[mpegts @ 0x7fe80a810a00] Packet corrupt (stream = 0, dts = NOPTS).
[mpegts @ 0x7fe80a810a00] Packet corrupt (stream = 0, dts = 1888887).
[h264 @ 0x7fe80b011800] concealing 846 DC, 846 AC, 846 MV errors in P frame
[mpegts @ 0x7fe80a810a00] Packet corrupt (stream = 0, dts = 2054052).
[mpegts @ 0x7fe80a810a00] Packet corrupt (stream = 0, dts = 2156154).
[mpegts @ 0x7fe80a810a00] Packet corrupt (stream = 0, dts = 2351349).
[mpegts @ 0x7fe80a810a00] Packet corrupt (stream = 0, dts = 2408406).
[mpegts @ 0x7fe80a810a00] Packet corrupt (stream = 0, dts = 2573571).
[mpegts @ 0x7fe80a810a00] Packet corrupt (stream = 0, dts = 2729727).
```

simspace-dev, 2021-04-06:
Hi @KonradIT, any chance you might know what's happening here and how I might be able to resolve this problem? I'm a developer. My Python skills are so-so, but I'd be happy to help if you can point me in the right direction of handling the "Packet corrupt" errors.  Seems like it may be an ffmpeg thing, but I'm not sure.

RolandSireyjol, 2021-07-08:
Hi @simspace-dev ,
I had a bit of a look since I had the same error on my end, with a GoPro Hero 8.
In the "stream" function, calling for "self.KeepAlive()" was causing the error.
By using the loop given in the example "opencv_gopro/ffmpegless-preview.py", I could recreate the same effect as we want from KeepAlive, but without the error. I also thing that it probably comes from ffmpeg itself.

here is the example file's content:
`#This is the script without the need of a FFmpeg installation, pure OpenCV
#This is not useful for image processing (eg: find faces) as there will be more lag, around 6 seconds added.
import cv2
import numpy as np
from time import time
import socket
from goprocam import GoProCamera
from goprocam import constants
gpCam = GoProCamera.GoPro()
sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
t=time()
gpCam.livestream("start")
cap = cv2.VideoCapture("udp://10.5.5.9:8554")
while True:
    nmat, frame = cap.read()
    cv2.imshow("GoPro OpenCV", frame)
    if cv2.waitKey(1) & 0xFF == ord('q'):
        break
    if time() - t >= 2.5:
        sock.sendto("_GPHD_:0:0:2:0.000000\n".encode(), ("10.5.5.9", 8554))
        t=time()
# When everything is done, release the capture
cap.release()
cv2.destroyAllWindows()`

RolandSireyjol, 2021-07-08:
@simspace-dev following on the **overview()** issue: I did get different errors depending on the kind of gopro I was setting up. Here are some various options:
>
    gopro = GoProCamera.GoPro(constants.gpcontrol)
    gopro = GoProCamera.GoPro(constants.auth)
    gopro = GoProCamera.GoPro()
  Best would be to make this function more resilient, but in the meantime what I did was making my own local **overview ** function, which is almost a copypaste of the lib"s version, without the crashing parts.