KonradIT/gopro-py-api #175: GoPro Hero9 control via USB issue

cristianilea-lateral, 2021-05-06:
I'm trying to control the webcam via USB (I got it to work via Wifi).
I tried a lot of things and I can't seem to get it to work via USB.
I should note that I've downgraded my firmware to v1.22.

Here is the ip of my Hero9:
[image]

I'm also aware of these tickets, and they did help to some extent (because I was doing some things wrong): #168  and #151.

Here is my code:

```
from goprocam import GoProCamera
from goprocam import constants

gopro = GoProCamera.GoPro(ip_address="172.27.180.54")
gopro.take_photo()
```

Here is my output of my code:
```
Waking up...
Waking up...
Waking up...
/bin/sh: vlc: command not found
Waking up...
Error, camera not defined.
```

Then I get an exception at the line `gopro.take_photo()`:
```
Exception has occurred: TypeError
argument of type 'NoneType' is not iterable
```

What should I try next?

KonradIT, 2021-05-06:
Hi, what version of gopro-py-api are you using?

cristianilea-lateral, 2021-05-06:
4.1.0 installed with pip

cristianilea-lateral, 2021-05-06:
Python is not my main programming language, so bear with me :D.

I went step by step with the pycharm debugger and it seems that it gets stuck somewhere in `whichCam`. This line throws an exception and then it goes downhill:
`response_raw = self._request("gp/gpControl")`

It builds the following uri `'http://172.27.180.54/gp/gpControl'`. If I try to make the request manually I get an exception like 
`urllib.error.URLError: <urlopen error [Errno 61] Connection refused>`

KonradIT, 2021-05-06:
Then it's not the correct IP... Have you tried with 172.27.180.51?

It's very important you look at the examples, [look how I did it](https://github.com/KonradIT/gopro-py-api/blob/master/examples/hero9-take-photo-webcam.py#L8), you pass an interface name to `getWebcamIP` as well as setting `webcam_device` to the interface name. Then it should work.

cristianilea-lateral, 2021-05-06:
With 172.27.180.51 it works... but why?

KonradIT, 2021-05-06:
Because the GoPro listens on the 51 octet of the IP.