KonradIT/gopro-py-api #178: Gopro Hero 8: change of resolution for streaming does not work.

RolandSireyjol, 2021-07-08:
**Describe the bug**
Using a Gopro Hero 8 from a macbook and calling for the `gopro.livestream("start")` runs, but only on low resolution. Changing the resolution using `gopro.streamSettings()` does not update the resolution.
I can change the camera FOV from the camera itself, and the resolution is either  (480, 640) or (480, 848), depending on the camera's settings, but never higher than that.

Attached to this post, a testing script showing the changes I've made to try and edit the resolution.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect via wifi to the gopro
2. Run the attached script
3. See that the resolution has not changed
4. Comment / uncomment the indicated lines of the script
5. See that the resolution has not changed

**Desktop:**
 - OS: macOS Big Sur
 - Version 11.2.3

**Additional context**
Note that it is not very resilient: I sometimes get ffmpeg errors forcing em to restart the system (missing headers...)
```python
import cv2`
import socket
import enum
from time import time
import subprocess
from goprocam import GoProCamera, constants

 def local_test():
     sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)

    gopro = GoProCamera.GoPro(constants.gpcontrol)  # pick one
    # gopro = GoProCamera.GoPro(constants.auth)       # pick one (this one does not work: for gopro 4)
    # gopro = GoProCamera.GoPro()                     # pick one

    t = time()
    gopro = change_gopro_params(gopro)
    gopro = stream(gopro, "udp://10.5.5.9:8554", "low")
    gopro = change_gopro_params(gopro)
    cap = cv2.VideoCapture("udp://10.5.5.9:8554")
    # width = 1920                              # tried to add that, didn't work
    # height = 1080                             # tried to add that, didn't work
    # cap.set(cv2.CAP_PROP_FRAME_WIDTH, width)  # tried to add that, didn't work
    # cap.set(cv2.CAP_PROP_FRAME_HEIGHT, height)# tried to add that, didn't work
    while True:
        _, frame = cap.read()
        print(frame.shape)
        cv2.imshow("GoPro OpenCV", frame)
        if cv2.waitKey(1) & 0xFF == ord('q'):
            break
        t = local_KeepAlive(sock, t)

    cap.release()
    cv2.destroyAllWindows()

def change_gopro_params(gopro):
    """
    tried to edit params this way, commenting/adding lines.
    didn't work too
    """
    gopro.video_settings("1080p")  # tried to add that, didn't work
    
    # gopro.gpControlSet(constants.Stream.IDR_INTERVAL, constants.Stream.IDRInterval.IDR4)
    # gopro.gpControlSet(constants.Stream.BIT_RATE, constants.Stream.BitRate.B250Kbps)
    # gopro.gpControlSet(constants.Stream.WINDOW_SIZE, constants.Stream.WindowSize.R720_1by2Subsample)
    # gopro.gpControlSet(constants.Stream.WINDOW_SIZE, constants.Stream.WindowSize.R720)
    # gopro.gpControlSet(constants.Photo.RESOLUTION, constants.Photo.Resolution.R12W)
    return gopro

def stream(gopro, addr, quality=""):
    """
    DISCLAIMER: this is an almost identical copy of the function gopro.stream()
    The only difference is that "KeepAlive" is commented, since it crashes the process.
    the function loc_KeepAlive does the same job, but was inspired from 
    "examples/opencv_gopro/ffmpegless-preview.py"
    """
    gopro.livestream("start")
    if gopro.whichCam() == constants.Camera.Interface.GPControl:
        if "HERO4" in gopro.infoCamera("model_name"):
            if quality == "high":
                gopro.streamSettings("2400000", "6")
            elif quality == "medium":
                gopro.streamSettings("1000000", "4")
            elif quality == "low":
                gopro.streamSettings("250000", "0")
        else:
            if quality == "high":
                gopro.streamSettings("4000000", "7")
            elif quality == "medium":
                gopro.streamSettings("1000000", "4")
            elif quality == "low":
                gopro.streamSettings("250000", "0")
            # subprocess.Popen("ffmpeg -f mpegts -i udp://" +
            #                  ":8554 -b 800k -r 30 -f mpegts " + addr, shell=True)
            # gopro.KeepAlive()
    # elif gopro.whichCam() == constants.Camera.Interface.Auth:
    #     subprocess.Popen("ffmpeg -i http://" +
    #                      "live/amba.m3u8 -f mpegts " + addr, shell=True)
    return gopro

def local_KeepAlive(sock, t):
    """
    Does the same job as gopro.KeepAlive, but was inspired from 
    "examples/opencv_gopro/ffmpegless-preview.py"
    """
    if time() - t >= 2.5:
        sock.sendto("_GPHD_:0:0:2:0.000000\n".encode(), ("10.5.5.9", 8554))
        t = time()
    return t

if __name__ == '__main__':
    local_test()
```

ozanpkr, 2021-08-13:
@RolandSireyjol how could you connect camera from laptop?Could you explain to me with details?

MihneaJurca, 2022-08-22:
I got the same problem did you manage to solve it?