KonradIT/gopro-py-api #180: Hero9 Black - Shoot video and sample scripts not working, device hangs up

faivala-de, 2021-09-03:
**Describe the bug**
a) Running the provided "download_video.py" of the repo leads to the camera not responding anymore. It's required to remove the battery for the device to restart.
b) Running shoot_video triggers the video recording, but recording does not stop.
c) Running shoot_video from an USB connection leads to the camera waking up, but no recording starts. 

**To Reproduce**

Following script leads to a video recording starting but not ending:

_from goprocam import GoProCamera, constants
import time
import sys

gpCam = GoProCamera.GoPro(ip_address=GoProCamera.GoPro.getWebcamIP(
    sys.argv[1]), camera=constants.gpcontrol, webcam_device=sys.argv[1])

videos_duration=[10,15,]
gpCam.video_settings("1080p","60")
gpCam.gpControlSet(constants.Video.PROTUNE_VIDEO, constants.Video.ProTune.ON)

for i in videos_duration:
	print("Recording " + str(i) + " seconds video")
	gpCam.downloadLastMedia(gpCam.shoot_video(9), custom_filename="VIDEO_aaa.MP4")
    
print("end")_

Script run with _python3 script.py 0_ (WLAN connection) leads to the camera recording but nothing happening.

If I connect to the camera via USB (Gopro Connect mode, get device name with ifconfig) the scripts will run without error but the camera will not react at all after waking up. Script then run as  _python3 script.py 0 en7_ 

en7: flags=8863<UP,BROADCAST,SMART,RUNNING,SIMPLEX,MULTICAST> mtu 1500
        options=6467<RXCSUM,TXCSUM,VLAN_MTU,TSO4,TSO6,CHANNEL_IO,PARTIAL_CSUM,ZEROINVERT_CSUM>
        ether aa:20:04:54:6f:a6 
        inet6 fe80::10e2:add0:532e:4654%en7 prefixlen 64 secured scopeid 0x1b 
        inet 172.26.140.54 netmask 0xffffff00 broadcast 172.26.140.255
        nd6 options=201<PERFORMNUD,DAD>
        media: autoselect (100baseTX <full-duplex>)
        status: active

**Expected behavior**
Recording starts and ends automatically. 
Camera should start recording (USB connection).

**Desktop (please complete the following information):**
Connect to camera via WiFi in Mac OS 11.3.1 , ARM M1. Firmware: v1.50.
Connect to camera via USB too.

KonradIT, 2021-09-03:
Note you cannot use shutter controls over WiFi or USB on current firmwares. Refer to this repository for information on older firmware compatibility.

faivala-de, 2021-09-03:
So I should try with Firmware 1.2x right? Vielen Dank für die schnelle Antwort!

KonradIT, 2021-09-03:
You'll still not be able to stop a recording, but you can start it over WiFi/USB on this FW.

For full control: https://github.com/konradit/gopro-ble-py

faivala-de, 2021-09-03:
Thanks. Now I can at least trigger the video recording start via USB, I have to stop it by clicking on the camera. 
Do you know if it's possible to control the camera both over BLE and USB at the same time? (first trigger start stop with BLE, then grab video via downloadLastMedia). This is especially for a multi camera setup interesting. Thanks a lot.

KonradIT, 2021-09-03:
No