KonradIT/gopro-py-api #183: GoPro Hero 8 USB live streaming url with opencv

BrainGym, 2021-10-02:
Hi, I wanted broadcast from GoPro Hero 8 firmware 2.50 using USB and display live with opencv.

I used [ifaddr interface](https://pypi.org/project/ifaddr/) to get the IP address (work fine)

`            for adapter in ifaddr.get_adapters():
                if "GoPro" in adapter.nice_name:
                    for ip in adapter.ips:
                        if ip.is_IPv4:
                            addr = ip.ip.split(".")
                            addr[len(addr) - 1] = "51"
                            addr = ".".join(addr)
`

then I used to IP address to wake up the gopro 
`gopro = GoProCamera.GoPro(ip_address=addr, camera=constants.gpcontrol)`

[image]

I want to get the live streaming by using livestream("start") function, but I lost here, I get an error when I pass the address in opencv 
`cam= cv2.VideoCapture(addr)
_, frame = cam.read()
cv2.imshow('Image', frame)`
error: (-215:Assertion failed) size.width>0 && size.height>0 in function 'cv::imshow'

I even tried with vlc with this url: http://IP_address/gp/gpControl/execute?p1=gpStream&c1=start
but nothing shows up

KonradIT, 2021-10-02:
You have to use a properly constructed UDP URI.

`udp://%s:8554 % addr`

BrainGym, 2021-10-02:
u mean by using gopro.stream("udp adress")  function?

KonradIT, 2021-10-02:
Nope. For stream function you have to provide your own addy.

BrainGym, 2021-10-04:
Finally I figure it out
After waking up the camera, I must call the startWebcam() function, then use an udp address,
it work fine with opencv

```
gopro = GoProCamera.GoPro(ip_address=addr, camera=constants.gpcontrol)
gopro.startWebcam()
udp_addr = "udp://@{}:8554".format(addr)
```

BrainGym, 2021-10-04:
Sorry @KonradIT  another question, After calling  the stopWebcam() function, the camera mantain on the ready mode is there in a function to put in normal mode or something ?

KonradIT, 2021-10-04:
What do you mean normal mode? You could try putting it in video/photo mode.

BrainGym, 2021-10-04:
For you question I mean the mode that display usb connected on the camera screen
putting it in video mode, l will try that

sakthivelj, 2022-11-20:
I need to connect two camera to raspberry Pi. When using alone it's working fine how can connect both. 

How can I switch the port? For one camera