KonradIT/gopro-py-api #184: GoPro 9. Disconnect on capture via USB

skit4214, 2021-10-09:
Hi

Trying to capture photo/video with GoPro 9 Black, connected via USB.
When start capture command gopro.shoot_video(10) or gopro.take_photo() camera disconnect from PC (USB switch off Windows sound) and reconnect after process. In take_photo case it reconnected after photo captured. In shoot_video it reconnect only after click capture button on camera and finish video recording

With WiFi behavior is the same. For video it start recording and can't finish by commands but without errors in log

With gopro.downloadLastMedia() disconnect not happens and all is ok

**Code for USB connection:**
```
from goprocam import GoProCamera

gopro = GoProCamera.GoPro(ip_address="172.20.163.51")
gopro.shoot_video(10)
#gopro.take_photo()
```

Log (same for photo and video):
```
HERO9 Black
HD9.01.01.22.00
Camera successfully connected!
Traceback (most recent call last):
  File "C:/Proj/[2021-10-04] Robovideo/GOPRO/GoProTest/GoProTest", line 4, in <module>
    gopro.shoot_video(10)
  File "C:\Proj\[2021-10-04] Robovideo\GOPRO\GoProTest\venv\lib\site-packages\goprocam\GoProCamera.py", line 601, in shoot_video
    self.shutter(constants.start)
  File "C:\Proj\[2021-10-04] Robovideo\GOPRO\GoProTest\venv\lib\site-packages\goprocam\GoProCamera.py", line 394, in shutter
    return self.gpControlCommand("shutter?p=" + param)
  File "C:\Proj\[2021-10-04] Robovideo\GOPRO\GoProTest\venv\lib\site-packages\goprocam\GoProCamera.py", line 168, in gpControlCommand
    return self._request("gp/gpControl/command/" + param)
  File "C:\Proj\[2021-10-04] Robovideo\GOPRO\GoProTest\venv\lib\site-packages\goprocam\GoProCamera.py", line 154, in _request
    return urllib.request.urlopen(uri, timeout=_timeout, context=_context).read().decode("utf-8")
  File "C:\Users\1\AppData\Local\Programs\Python\Python37\lib\urllib\request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
  File "C:\Users\1\AppData\Local\Programs\Python\Python37\lib\urllib\request.py", line 525, in open
    response = self._open(req, data)
  File "C:\Users\1\AppData\Local\Programs\Python\Python37\lib\urllib\request.py", line 543, in _open
    '_open', req)
  File "C:\Users\1\AppData\Local\Programs\Python\Python37\lib\urllib\request.py", line 503, in _call_chain
    result = func(*args)
  File "C:\Users\1\AppData\Local\Programs\Python\Python37\lib\urllib\request.py", line 1347, in http_open
    return self.do_open(http.client.HTTPConnection, req)
  File "C:\Users\1\AppData\Local\Programs\Python\Python37\lib\urllib\request.py", line 1322, in do_open
    r = h.getresponse()
  File "C:\Users\1\AppData\Local\Programs\Python\Python37\lib\http\client.py", line 1344, in getresponse
    response.begin()
  File "C:\Users\1\AppData\Local\Programs\Python\Python37\lib\http\client.py", line 306, in begin
    version, status, reason = self._read_status()
  File "C:\Users\1\AppData\Local\Programs\Python\Python37\lib\http\client.py", line 267, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "C:\Users\1\AppData\Local\Programs\Python\Python37\lib\socket.py", line 589, in readinto
    return self._sock.recv_into(b)
ConnectionResetError: [WinError 10054] Удаленный хост принудительно разорвал существующее подключение

Process finished with exit code 1
```

Tried with commands in browse with video:
```
http://10.5.5.9/gp/gpControl/command/shutter?p=1 //video start recording
http://10.5.5.9/gp/gpControl/command/shutter?p=0 //noting
```

**Versions:**
GoPro 9 Black downgraded to 1.22
Goprocam-4.2.0
Windows 10

skit4214, 2021-10-09:
Downgraded to 1.21 
Same behavior

gritzandgravy0, 2022-02-09:
Hey,

Did you ever figure this out?  I'm getting the same error.

Thanks,

Joe

dyldata, 2022-04-06:
@gritzandgravy0 @skit4214 resurfacing this - you two have any luck here?

skit4214, 2022-04-14:
@dyldata no, i was reading many materials about this problem and it's not real do for GoPro9 by software. If necessary only for remote capture control GoPro9 i found only one way - disassemble the device and add device (like arudino) in chain. It will lock and unlock capture button. In result i skipped GoPro9 and make software with other camera (
240 frames was the only condition)

Them