KonradIT/gopro-py-api #186: Streaming from multiple cameras

joshi-bharat, 2021-11-08:
I am trying to stream from 2 cameras as stereo cameras. I am trying to collect data in real-time. I tried to use the cameras as a webcam stream.

The problem is both the cameras stream on the same port. Hence, the signals are mixed and corrupted. 

Has anyone tried this or know work around this? Any hints are highly appreciated.

joelemanoel, 2022-02-11:
@joshi-bharat any solution?

KonradIT, 2022-02-13:
Hi, this is currently a wontfix since it requires external hardware. A way to fix it is using a separate device to connect to the GoPro (eg: a raspberry pi) and issue a different port on the device.

joshi-bharat, 2022-02-13:
> @joshi-bharat any solution?

Yes, I was able to stream two GoPros using port forwarding.  For instance, if you have another gopro with server address 172.22.186.51 and local address 172.22.186.54. Then, you can do port forwarding as follows

```
sudo iptables -t nat -A PREROUTING -p udp -s 172.22.186.51 --dport 8554 -j DNAT --to-destination 172.22.186.54:23002 
```
I also tested as follows:
Computer A publishing as follows
```
ffmpeg -stream_loop 500 -i sample.mp4 -vcodec mpeg4 -f mpegts udp://<target ip address>:23001
```
computer B publishing as follows
```
ffmpeg -stream_loop 500 -i sample2.mp4 -vcodec mpeg4 -f mpegts udp://<target ip address>:23001
```
Computer C (target ip address) receiving both

```
ffplay udp://<target ip address>:23001 # one terminal

# in another terminal
sudo iptables -t nat -A PREROUTING -p udp -s <IP address of computer B> --dport 23001 -j DNAT --to-destination <target IP address>:23002 
ffplay udp://<target ip address>:23002 # second terminal
```

closed 2022-03-04 by joshi-bharat