KonradIT/gopro-py-api #188: Gopro Hero 9 Won't Capture Videos or Photos

gritzandgravy0, 2022-01-14:
Hey,

So the API has been great and easy to setup, so thanks for that. 

It seems as though all the functions that I call out, except .take_photo or .take_video, work.  I can connect to the camera no problem, download media from it onto my PC, pull information from the gopro, power off, etc.  When I try to run either the .take_photo or .take_video the screen on the gopro will start strobing in and out of brightness for a few seconds then the red light will blink and the rear screen shows a capture of a picture and has a little animation of it moving down on the screen.  The problem is that the image or video never actually saves in the memory of the SD card whether I try to pull it using python or just checking manually.  Also to mention there are no errors in python or anything that indicates that anything went wrong, it will complete the script.  When I take a photo manually, pressing the red button on top, there is none of that strobing that I see when using python, it's just the little animation of the image moving down on the screen and the red LED.  

The kicker is that when I connected up my gopro hero 5 session instead everything works flawlessly including taking photos and videos.  So I don't think I'm doing anything wrong....I'm wondering if my firmware version or python version isn't compatible.

I am currently using the following:
Python 3.10.1
Gopro firmware  HD9.01.01.60.00
Windows 10

I see that you mentioned this would work on the hero 9.  What firmware version was on that?

Thanks,

Joe

KonradIT, 2022-01-14:
Hi, please downgrade to 1.21 or 1.22

https://github.com/KonradIT/goprowifihack/blob/master/HERO9/HERO9-Functionality-Compatibility-Chart.md

gritzandgravy0, 2022-01-15:
Great thanks for the quick reply. I will try this out Monday. 

Joe

gritzandgravy0, 2022-01-17:
I downgraded to 1.22 and that worked like a charm. 

Thanks

closed 2022-02-10 by gritzandgravy0