KonradIT/gopro-py-api #196: Connect ubuntu and gopro10 with USB

yangshubin27400430, 2022-07-01:
Hello!
I want to use USB to control the camera
I also see the following solutions
[https://github.com/gopro/OpenGoPro/discussions/146](url)

According to its steps, I can now use the win10 to control the camera
But when using ubuntu again I encountered the same mistake as 146
Don't know how to set the Linux kernel device identifier

Run ls /dev/* 
Results of the command
dev/libmtp-1-2.1

I wonder if this problem has been successfully solved in the end?
(discussions/146)

We are looking forward to your reply
Thank you very much

yangshubin27400430, 2022-07-01:
ps:
The win10 is successfully connected to the camera(gopro10)
Is it because the driver is installed inside??
https://community.gopro.com/s/article/GoPro-Webcam?language=en_US

But this driver is only available in Windows and Mac versions

shahla-ai, 2023-12-04:
hello @yangshubin27400430 
any solution found to control vis USB on ubuntu ?