KonradIT/gopro-py-api #197: USB connection with a GoPro

mat-ver, 2022-07-18:
Greetings,

I am willing to use the GoProCam API to control a GoPro HERO 9 but I would like to do so through USB. Not particularly tech saavy here and I couldn't find this information directly: to use GoProCamera.getWebcamIP() I should connect the camera to computer using an USB-C/Ethernet converter, right? I've suceeded in controlling the camera through Wi-Fi, but I don't know if to use it with USB I should connect the USB directly to the computer or get an ethernet adapter. I have a feeling I must be missing something very essential.

The results I've gotten by connecting it directly are the IP 10.5.5.9 and "waking up...", without the connection confirmation. When I imagine the IP we get should be X.Y.Z.51.

Kind regards.

bosc0, 2022-07-18:
You have to connect the gopro without the ethernet adapter, just usb

mat-ver, 2022-07-18:
The entire process should be, then:

- turning camera on;
- setting USB connection as "GoPro Connection";
- connecting the GoPro to the computer via USB;
- attempting to connect to the camera through the right commands in the code.

Right?

I guess I will have to try to downgrade the firmware as the documentation suggests (currently using 1.60 and result is what is described above) and see if it works then, assuming this is right. I will return to this thread and share my results once I do it. Thank you for the answer, it is immensely helpful.

KonradIT, 2022-07-18:
Right.

mat-ver, 2022-07-21:
Thanks guys, managed to obtain video from the GoPro through USB thanks to you both. Just stating that I was unable to downgrade from 1.60 once both 1.21 and 1.22 ended making my camera virtually useless because the back screen goes off, but using 1.60 I still managed to get the image feed I needed (and just installing 1.60 again was enough to fully restore the GoPro's functionalities, in case someone who ever reads this goes through the same issue). That was very helpful!

Trying to make the best use for the thread before having it closed: to power the camera on through a command my only path would be via BLE, and my only path of using BLE would be by using gopro-ble-py, right? Having this webcam stream through USB and the ability of powering the camera on remotely (an alternative would be the operator simply using the Quik app to power on but we'd like to avoid such a solution) would pretty likely be all I need for this project.

Thanks again!

closed 2022-07-26 by mat-ver