KonradIT/gopro-py-api #200: Hero 10 webcam python script

druckgott, 2022-10-02:
Hallo I´m working on a script and want to read the hero 10 webcam video, but I do not want to show it with vlc player like it´s implementet, I want to use cv2

But I can´t get it working. The connection works I connect to the camera the hero 10 shows that webcam is aktiv, but I do not get a stream.

I did the same over wifi, therfore I get the stream but it was to laggy

Can you help me?

```
import cv2
from time import time
import socket
from goprocam import GoProCamera, constants
import queue
import threading

gpCam = GoProCamera.GoPro(ip_address="172.22.141.51")
sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
t=time()
gpCam.livestream("start")
gpCam.video_settings(res='1080p', fps='30')

q=queue.Queue()

def Receive():
    print("start Reveive")
    cap = cv2.VideoCapture("udp://172.22.141.51:8554", cv2.CAP_FFMPEG)
    while True:
        ret, frame = cap.read()
        if not ret or frame is None:
            print('video is all read')
            break
        q.put(frame)
    
    cap.release()
    cv2.destroyAllWindows()

def Display():
    print("Start Displaying")
    t=time()
    while True:
        if q.empty() !=True:
            frame=q.get()
            cv2.imshow("GoPro OpenCV", frame)
        if cv2.waitKey(1) & 0xFF == ord('q'):
                break
        if q.empty() != True:
            frame = q.get()
            # Save the picture 
            cv2.imshow("GoPro OpenCV", frame)
            # cv2.imshow(save_name, frame)
            if cv2.waitKey(1) & 0xFF == ord('q'):
                break
            if time() - t >= 2.5:
                sock.sendto("_GPHD_:0:0:2:0.000000\n".encode(), ("172.22.141.51", 8554))
                t=time()

if __name__=='__main__':
    p1 = threading.Thread(target=Receive)
    p2 = threading.Thread(target=Display)
    p1.start()
    p2.start()
```

KonradIT, 2022-10-02:
I fixed your code a little bit, now it works... kinda, it has a memory leak.

```python
camId = "{CF7CD38D-B0DE-4C2F-B190-7E3A21CA75AE}"
gpCam = GoProCamera.GoPro(ip_address=GoProCamera.GoPro.getWebcamIP(
    camId), camera=constants.gpcontrol, webcam_device=camId, api_type=constants.ApiServerType.OPENGOPRO)
sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
gpCam.startWebcam()
time.sleep(2)
q = queue.Queue()

def handler(s, f):
    p1.stop()
    p2.stop()
    p3.stop()
    quit()

signal(SIGINT, handler)

def Receive():
    print("start Receive on %s" % GoProCamera.GoPro.getWebcamIP(camId))
    cap = cv2.VideoCapture(
        "udp://%s:8554" % GoProCamera.GoPro.getWebcamIP(camId), cv2.CAP_FFMPEG)
    while True:
        ret, frame = cap.read()
        if not ret or frame is None:
            print('video is all read')
            break
        q.put(frame)

    cap.release()
    cv2.destroyAllWindows()

def Display():
    print("Start Displaying")
    while True:
        if q.empty() != True:
            frame = q.get()
            cv2.imshow("GoPro OpenCV", frame)
        if cv2.waitKey(1) & 0xFF == ord('q'):
            p3.stop()
            break
        if q.empty() != True:
            frame = q.get()
            # Save the picture
            cv2.imshow("GoPro OpenCV", frame)
            # cv2.imshow(save_name, frame)
            if cv2.waitKey(1) & 0xFF == ord('q'):
                break

if __name__ == '__main__':
    p1 = threading.Thread(target=Receive)
    p2 = threading.Thread(target=Display)
    p3 = threading.Thread(target=gpCam.KeepAlive)
    p1.start()
    p2.start()
    p3.start()
```

[image]

druckgott, 2022-10-02:
very nice, I did use it, hero 10 says aktive, but there is no windows apearing I get:

Waking up...
Camera successfully connected!
Connected to 172.22.141.51    
start Receive on 172.22.141.51
Start Displaying

KonradIT, 2022-10-02:
Make sure you use the code as-is and only change the `camId`, with your equivalent network interface id.

closed 2022-10-02 by KonradIT

druckgott, 2022-10-02:
And where can I get this camId in windows?

KonradIT, 2022-10-02:
Use `./examples/get_iface_name_win.py`

druckgott, 2022-10-03:
Ok I now get the ID but it´s not working changed it to:
camId = "{40BAB5DA-3918-43C7-98CF-55B4D403C704}"

Then If I use your code 1:1
I get an error:

NameError: name 'SIGINT' is not defined

This is what I import:

```
import cv2
import socket
from goprocam import GoProCamera, constants
import queue
import threading
import signal
import time
```

I have to Update the line:
```
signal(SIGINT, handler)
```
to
```
signal.signal(signal.SIGINT, handler)
```

Then I get no error, but its always the same: (Camera itself switch to webcam mode and says aktiv)
Waking up...
Camera successfully connected!
Connected to 172.22.141.51
Start Displaying
start Receive on 172.22.141.51
video is all read

No window is getting up.

Sorry for asking so much ;-)