KonradIT/gopro-py-api #202: Gopro Hero 11 support

Bartelime, 2022-10-30:
Dear all
I was using the library for a while with my GPH10. Recently I received my new GPH11 and unfortunately the gopro-py-api does not work anymore, like with GPH10. Does someone know what the difference in the firmware is?
I would appreciate, if someone could help getting the gopro-py-api for Gopro Hero 11 compatible. 

Thanks for your help
Regards
Bartelime

KonradIT, 2022-10-30:
Weird, I updated it to work with the Hero11, what exactly does not work?

Bartelime, 2022-10-30:
Hi there
tried to record video , it says, waking up.... until I touch the touchscreen from GPH11. then it suddenly starts recording, but it never ends.....
see picture with cmdl log
regards
Bartelime

Bartelime, 2022-10-30:
Hi there
just tried to list media overview, it has some str error about the getStatus --> see picture with code an error log
Regards
Bartelime

djpirra, 2023-01-12:
No updates regarding GoPro Hero 11 support? :( Sad

KonradIT, 2023-01-12:
The library will likely not receive any new updates. Since it doesn't support talking to the cam via Bluetooth, it can't work reliably with HERO8 and newer cameras. Switch to OpenGoPro.

https://medium.com/@konrad_it/sunsetting-my-unofficial-gopro-wireless-hacking-effort-a06a03c7fc1c

closed 2023-01-12 by KonradIT