KonradIT/gopro-py-api #208: Help: Control settings via usb

UnNolan, 2023-05-24:
Helloo!

First of all thank you for this amazing API!

I use Gopro Hero 10 connected by USB, and I take photos not videos. Is it possible to control the gopro settings with a wired connection? I would like to be able to configure the white balance, the iso, the field of view (linear, wide)... In short, all the protune parameters. And everything via usb. Is it possible? And if so, how ?

I tried with constants.py and gpControlSet but without success, maybe I didn't do it well

 Thanks a lot in advance for your help!

Nolan

UnNolan, 2023-11-30:
Helloo!
Yes I found a solution to control almost every settings, wich camera are you using ? :)

shahla-ai, 2023-12-04:
hello @UnNolan 
i am using **GoPro hero 12 black** on **linux**
the Cam is not showing on my /dev/tty list when connected via USB
any Help ?

shahla-ai, 2023-12-12:
@UnNolan Any Help ?