KonradIT/gopro-py-api #210: Control GoPRo hero 7 black via usb wired connection

shahla-ai, 2023-11-30:
hello @KonradIT 
i'm trying to control my GoPro Hero7 black via usb wired connection on my RaspberryPI
i followed the instructions on this issue [https://github.com/gopro/OpenGoPro/discussions/146](url)
the problem is that i'm not finding the **uuid** of my cam in /dev/* , or in other words nothing appears in my /dev/ when connecting my Cam
is there any thing i should do ?
best regards

KonradIT, 2023-11-30:
Hero7 doesn't support USB Ethernet

shahla-ai, 2023-11-30:
@KonradIT 
Thank you for your reply 
is there any way to get live stream or control my GoPro Hero 7 using wired USB cable ?
**Another Question Please**
My GoPro is not apearing in my WIFI available networks in my PI, but i turned ON my WIFI in the camera
is there anything i should do before ? 
Thank You

KonradIT, 2023-11-30:
> is there any way to get live stream or control my GoPro Hero 7 using wired USB cable ?

No